Lakemonsters2635/MonsterDocs #13: Index - Needs work

jwaltner, 2024-10-21:
the `index.rst` needs some work so that we have sections and the logical groups have group headings.  Someone needs to head-up learning RST format and getting us a good start on MonsterDocs formatting.

jwaltner, 2024-10-26:
Also please figure out how we do previews of these docs so we do not need to develop off of `main` to preview.  @venomboss985 , just tagging you for awareness on this, if you know how to do this, please let us know.

SahanRULES, 2024-10-27:
For preview of docs use a pre-release version (you can change this in vs code extensions) of the esbonio extension https://docs.esbon.io/en/latest/. Also, there needs to be a python interpreter integrated into the vs code project. It does not seem to work with the readTheDocs theme, however by publishing the page on github we can always check.

SahanRULES, 2024-10-27:
The section breaks have been added on the branch sectionBreaks which is of dev. The problem was the spacing because the caption does not appear if there is a line break in between the :maxdepth: and the :caption: 

[image]
[image]

jwaltner, 2024-10-31:
@SahanRULES , thank you for figuring out how to fix the index.  This makes the docs a lot better.  I have gone ahead and merged your updates into the main and that automatically built the new version of the docs.  I will checkout that preview as well.

jwaltner, 2024-10-31:
Item moved to In Review while I review the preview item.  Thanks for that as well.